sticilface/ESPmanager #3: Did you get platformio working with esp8266/Arduino stage?

eyp, 2016-02-01:
Hi, I'm sorry but I don't know how to get in touch to you.
I'm trying to get platformio working with 2.1.0-rc2 esp8266/Arduino sdk, but I can't get it working.
I was reading this thread: 
https://github.com/platformio/platformio/issues/401

So I've followed all the instructions and added -lmesh to the platformio.ini config file, but I'm still getting the error:

/Users/edu/.platformio/packages/framework-arduinoespressif/tools/sdk/lib/libcrypto.a(aes-internal.o):(.irom0.text+0x0): undefined reference to `rcons'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/esp12e/firmware.elf] Error 1

Did you solve the issue? Would you mind to tell me how? (eduardo.yanez@gmail.com)

Thanks in advance.

chaton78, 2016-02-02:
I've fixed that issue by adding 
`PROVIDE` ( rcons = 0x3fffd0f0);
to esp8266.flash.common.ld in .platformio\packages\ldscripts

eyp, 2016-02-02:
Thank you!!!!
And how did you get the info to do that?, because to be honest, I'm a complete newbe about .ld files and esp8266 linking.

sticilface, 2016-02-02:
It was My understanding that this was all fixed in later Versions of Platformio. I did not have to add the rcons myself to the ld  script.

chaton78, 2016-02-02:
It was under my impression too but I started a fresh install of Platformio
yesterday. Followed the instructions from the issue 401, still having the
issue (is it me or something else,  not sure:) ). We know what to do for
future release of the SDK....

sticilface, 2016-07-20:
This v2.0.0-rc1 works with the standard lib from platformio now.  Hope that helps

sticilface, 2016-11-13:
http://docs.platformio.org/en/stable/platforms/espressif8266.html#using-arduino-framework-with-staging-version

sorted

closed 2016-11-13 by sticilface