sticilface/ESPmanager #6: no match for 'operator=' (operand types are 'ArduinoJson::JsonObjectSubscript<const String&>' and 'FlashMode_t')

EagleSmith, 2016-04-05:
hi,it's not working 
my arduino ide is 1.6.8 ,sdk is 2.1.0
sorry for my english,can you help me ?thanks

C:\Users\Eagle\Documents\Arduino\libraries\ESPmanager-0.1\ESPmanager.cpp:1395:38: error: no match for 'operator=' (operand types are 'ArduinoJson::JsonObjectSubscript<const String&>' and 'FlashMode_t')

         root[F("flashchipmode_var")] = ESP.getFlashChipMode();

sticilface, 2016-04-05:
What's your version of arduinojson?

sticilface, 2016-04-05:
Looking at the esp code this function returns a flashtype enum. 
Try changing it to this 
'''
     root[F("flashchipmode_var")] = (uint8_t)ESP.getFlashChipMode();
'''

sticilface, 2016-04-05:
Looking at it, this was actually fixed in commit 
4de0153ee7244f3767b4e5e80e692c51083767c8

I suggest you get the latest version of ESPmanager, and see if it persists.

EagleSmith, 2016-04-07:
ok,it's working now ,my arduinojson is ArduinoJson-5.1.1

closed 2016-04-07 by EagleSmith