sticilface/ESPmanager #7: Add GPIO, NTP, MQTT

adimavi, 2016-04-28:
Hi,
Nice work you can improve the ESPmanager by adding GPIO, NTP, MQTT and Automation uses NTP, OTA by own ESP. That will be amazing.

sticilface, 2016-04-28:
haha... that would be good.. I had thought about MQTT, but left it out as i figured people would want to control their sketch with it, and that would save having two instances!  

There is the AREST lib, which is good for controlling just the GPIOs.

Can you give me a bit more detail on how u want to use these features?

adimavi, 2016-04-29:
Visual on/off buttons for toggle GPIO,
NTP for get the time from internet for do the own automation with one esp - like irrigation control - turn on the gpio in that time turn off in that time
For basic mqtt code i create a gist https://gist.github.com/adimavi/fcb4f1d8d8d67f76ddac2e1106d9f904

Then the ESP do its own automation and for the genaral automation it can use the mqtt broker easily.

sticilface, 2016-05-01:
Ok. I get the idea. That is definitely a good idea.  I'm not sure if want to implement it as a feature of espmanager though. The reason being that I feel all that functionality is better of being part of the main sketch.  The main sketch takes control of the /root directory and offers everything that you've mentioned. What I can offer is at some point I'll make an example that does that? How does that sound?

closed 2016-12-10 by sticilface