sticilface/ESPmanager #8: http basic auth

martindb, 2016-06-27:
Can you add http basic auth to the setup area?

sticilface, 2016-06-28:
I have been thinking about auth / security for a little bit.  

I think that there will be some fairly big changes to the IDE in the not too distant future to allow authentication of binaries and other security.  

The thing that has stopped me is how to implement this. 

1) hard coded into sketch 
2) changeable in GUI
3)  where to store them ? hash in the SPIFFS settings file?  

Do you have any suggestions? 

PS.  there have been a load of changes to this.  in the gui branch.  moved to async web server,etc.  which is much better.  web sockets. updating via a json file, including binary, with continued progress shown by the web sockets.  you can now subscribe to a url, and pull all changed spiffs files, and new binary if different to current running one, with a user defined checking frequency. and a lot of bug fixes that i don't seem to be able to merge into the sync branch..

martindb, 2016-06-29:
I think that a "default" hardcoded password is ok for easy starting.
With GUI you can change this default, and it's a good idea I think to store the hash in the settings file. Another option can be the eeprom, but, at the end, it's in the same flash memory... in the json file is more easy to deal with it.

I was trying your code in the master, and viewing the gui branch. I think you have to switch the master to the async version. Sounds great the "auto update" feature!
I'll try the gui branch version in the next days.

sticilface, 2016-07-15:
Ok.. so this is on hold for a little bit... there needs to be some work on the back end of authentication in asyncwebserver.  ESPmanager uses an index file, 3 js files, 1 css file and a data.esp which is the handler function.  These are handled by different handlers.. so managing authentication between them in a seamless way is not possible yet...  

it will be.  me-no-dev is working on it.  so will update then.  hope that is ok.  

In the mean time.  I'm working on a total rewrite which should be a lot better.  It is already and has some better features. not pushed any changes yet!

martindb, 2016-07-16:
Sounds great!!! Thank you very much.

sticilface, 2016-12-10:
this is still in the plan... just waiting for the implementation in asyncwebserver

sticilface, 2018-05-01:
coming in version 3.  may add though