sticilface/ESPmanager #10: compilation problems

leonardpitzu, 2016-11-08:
problem #1 - DEBUG cauzez problems with arduino.h
problem #2 - after fixing the above issue ESPmanager.cpp:202:10: error: no matching function for call to 'ArduinoOTAClass::onStart(ESPmanager::begin()::__lambda0)'
         });

i use latest versions of arduino ide/compiler/libraries + esp8266 sdk/arduino core

sticilface, 2016-11-08:
can u post the full error code?   bit weird as I'm not having any problems compiling with arduino

leonardpitzu, 2016-11-09:
it was your sample which was not working and i posted the code after extensive testing. apparently not extensive enough though as an uninstall, reboot and then fresh install of arduino + esp8266 sdk/compiler/arduino libs resolved the issue. thanks for your quick reaction and sorry for the false report.

sticilface, 2016-11-13:
no problem.

closed 2016-11-13 by sticilface