sticilface/ESPmanager #11: Document how to integrate with own sketch

probonopd, 2016-12-04:
Let's say I want to write a sketch that checks my GitHub Notifications regularly, and lights a LED if there are some.

For this, I would need to somehow enter my GitHub token. I don't want this to be hardcoded in the sketch but to be entered by the user into ESPmanager.

It would be nice if there was a super simple way to pass in the needed variable, and (auto)generate a GUI entry field for it.

Can you show how to do this please?

sticilface, 2016-12-04:
I'm not sure how to achieve that.   I'd not want to integrate it into the espmanager, as it sounds more like the job of the main sketch.  ESPmanager is designed to slot in with any sketch and to manage the wifi / AP / OTA stuff.... 

The way i get things working would be to make a sketch that does what you are describing, use asyncwebserver to host the page you are describing on the root of the esp... i.e. /index.htm then just include espmanager as is done in the example to handle the wifi stuff.  If you are after a page that looks the same then the code is in the www folder.  you should be able to run the gulp script to generate the page, the script is in there! 

You'd have to get ssl working though, as i think everything through git is https..

does any of that make sense? or have i misunderstood you?

probonopd, 2016-12-04:
The functionality should be implemented in the main sketch for sure. But it would be cool if the main sketch could tell the ESPmanager "give me the value for variable $TOKEN". Maybe a configuration file on the filesystem could be used to define a list of variables that ESPmanager would ask the user to input, and feed them to the sketch.

I hope you understand what I am looking for; if not, let me know and I'll elaborate.

sticilface, 2016-12-07:
I can have a look at something like that.  I've done it for the Melvanimate stuff... variables for effects.  
The issue might be code bloat.. as it might end up taking a lot of ram.  especially with strings. 

I think that code injection is out...  maybe...  although i just thought of an idea that would work!  might be worth a try... have to be client side injection.  Would have to be careful though as if you define lots of strings with the html to inject, this will all take RAM.

probonopd, 2016-12-07:
Maybe use config files in spiffs?

sticilface, 2018-05-01:
not really made work with this.. im rewriting a version 3 for ESP32 and ESP8266 can try to include it there.  very slow progress though.

closed 2018-05-01 by sticilface