sticilface/ESPmanager #12: Reduce number of includes needed in sketch

probonopd, 2016-12-05:
Right now we need to

```
#include <FS.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncJson.h>
#include <ArduinoOTA.h>
#include <ArduinoJson.h>
#include <Hash.h>
#include <ESP8266mDNS.h>
#include <ESPmanager.h>
```

Wouldn't it be possible for ESPmanager.h to include the others, so that the main sketch could stay a bit leaner?

sticilface, 2016-12-05:
ESPmanager does include all these, but these are all seperate libs... except maybe the hash.h lib, and from what i understand of the way the the arduino system works, they have to be included in the main sketch for the build process to find them... ok so i thought i'd check that and it seems like within arduino you can now ditch them all!  and just use ESPManager.h

the gotya however is that this does not work for platformio (which i use predominately).  so i would be keen to keep them there.  but if u can get it work with less be my guest.  They are all included in the appropriate places in the lib, so should work.

probonopd, 2016-12-05:
Ah, good to know. Maybe we should ask platformio why it behaves differently from Arduino? In the ideal world, platformio could run all Arduino sketches unchanged...

sticilface, 2016-12-05:
It used to be like that. But I'm using latest IDE and it seems to work. I'll ask pio....

It is useful though as it shows u a list

tablatronix, 2017-10-09:
A little late , i think Platformio requires you specify lib linking eg.  Lib_ldf_mode deep

sticilface, 2018-05-01:
think this is resolved now.  can just include the main espmanager header

closed 2018-05-01 by sticilface