sticilface/ESPmanager #13: Known network not in list

bertmelis, 2016-12-09:
I just tested the example in the ESPmanager but didn't make it through the complete configuration.

I have a lot of Wifi networks in the neighbourhood and since the list of possible networks is rather "short", I simply cannot connect to my own network.

I see 2 possible workarounds: 1) make the list longer or 2) add a input filed to type the SSID of the known network.

Or am I missing something?

sticilface, 2016-12-10:
This is a very good point. I will add manual ssid and pass. This will hopefully be fixed in my next release. 

Currently I'm limiting the number of wifi networks found to 10 as I was getting a lot of crashes. I think I've fixed that now so I can probably increase it. But I'm also part way through adding sorting so it list them all strongest to weakest. Which will also help. Hopefully by sun. 
A

probonopd, 2016-12-10:
I also only see 10 networks and my netwrok is not in this list. Please make it possible to show all of them, e.g., on more than 1 page...

sticilface, 2016-12-10:
yes that was intentional.  I was getting a lot of crashes and thought it might be due to the buffer getting too big with more than 10 networks. the json ends up massive, and i'm meaning to reduce it.  however, i later tracked the instability to the way in which i was fetching the data.  I was leaving the request open instead of getting the client to check if it was done.  The request was getting closed occasionally which is what was causing the crashes.  I've changed the client, and now its all good. 

https://github.com/sticilface/ESPmanager/commit/8cf5007e1d864e78794884372af384aa4022301e

This commit adds back a limit of 20 networks, and also sorts it so that the strongest appear first.  
Should solve this issue. 

I'll add manual ssid later.

closed 2016-12-10 by sticilface