sticilface/ESPmanager #14: Cannot use "default hard coded set of credentials"

probonopd, 2016-12-10:
When trying to use `You can specify a default hard coded set of credentials`:

```
ESPmanager-example:38: error: no matching function for call to 'ESPmanager::ESPmanager(AsyncWebServer&, fs::FS&, const char [11], const char*&, const char*&)'
 ESPmanager settings(HTTP, SPIFFS, "ESPManager", defaultSSID , defaultPSK);
                                                                         ^
/tmp/arduino_modified_sketch_730001/ESPmanager-example.ino:38:73: note: candidates are:
In file included from /tmp/arduino_modified_sketch_730001/ESPmanager-example.ino:27:0:
/home/me/Arduino/libraries/ESPmanager-master/src/ESPmanager.h:92:9: note: ESPmanager::ESPmanager(AsyncWebServer&, fs::FS&)
         ESPmanager(AsyncWebServer & HTTP, FS & fs = SPIFFS);
         ^
/home/me/Arduino/libraries/ESPmanager-master/src/ESPmanager.h:92:9: note:   candidate expects 2 arguments, 5 provided
/home/me/Arduino/libraries/ESPmanager-master/src/ESPmanager.h:89:7: note: ESPmanager::ESPmanager(const ESPmanager&)
 class ESPmanager
       ^
/home/me/Arduino/libraries/ESPmanager-master/src/ESPmanager.h:89:7: note:   candidate expects 1 argument, 5 provided
exit status 1
no matching function for call to 'ESPmanager::ESPmanager(AsyncWebServer&, fs::FS&, const char [11], const char*&, const char*&)
```

sticilface, 2017-01-09:
I've actually removed this now.   It was proving too complicated.  It seemed like a good idea at the start as it allowed a quick config... but then you end up with setting some in software later and not knowing which ones to override etc...

closed 2017-01-09 by sticilface

probonopd, 2017-01-11:
OK, thanks for the clarification and cleanup.