sticilface/ESPmanager #15: Easy way to delete messed up settings

probonopd, 2016-12-10:
Looks like I have managed to mess up my WLAN settings with the result that I seem to be unable to get back in at all. Would be great if there was an easy way to reset all spiffs settings, e.g., by shorting a certain pin on the ESP.

sticilface, 2016-12-10:
it should not really do that.  hopefully not anyway, as it has to successfully join the network and return true, otherwise it sets up an access point.   

I do see your point thought.  I would recommend one of the following... 

1) re-upload the SPIFFS data folder from the example.  this will reset it. 
2) call ```settings.factoryReset()``` from your sketch which you can do just once.. 
3)  add your own handlers in the main sketch to then call factory reset...

I've done 3 for another project... 

where gpio 0 does a bunch of different functions. it relies on the Bounce2 lib.  This code does a whole bunch of stuff...  a short press toggles the value of one pin, a longer press causes a reset, and an even longer one causes a factory reset.  i use it on my SONOFFs devices, and it works perfectly. 

I still think that this sort of functionality is best handled in the actual user sketch.. 

```
    // Update the Bounce instance :
    bool changed = debouncer.update();

    if ( changed ) {
        // Get the update value
        int value = debouncer.read();
        if ( value == LOW) {

            buttonPressTimeStamp = millis();

        } else {

            if (millis() - buttonPressTimeStamp < 500) {
                output(!pin_state);
            }

            if (millis() - buttonPressTimeStamp < 10000 &&  millis() - buttonPressTimeStamp > 2000) {
                _Debugf("\nRESTARTING....\n");
                output(false);
                delay(2000);
                digitalWrite(STATUS_LED_PIN, HIGH);
                ESP.reset(); 
            }

        }
    } else {

    if (millis() - taskTimeout > 100) {
        taskTimeout = millis(); 

        if (debouncer.read() == LOW) {

            //  creates flashhing status LED to indicate you're about to factory reset. 
            if (millis() - buttonPressTimeStamp > 2000 && millis() - blinkTimeout > 500) {
                digitalWrite(STATUS_LED_PIN, !digitalRead(STATUS_LED_PIN));
                blinkTimeout = millis(); 

            }

            if (millis() - buttonPressTimeStamp > 10000) {
                _Debugf("FACTORY RESET....\n");
                output(false);
                digitalWrite(STATUS_LED_PIN, HIGH);
                settings.factoryReset();
                SPIFFS.remove(MY_SETTINGS_FILE);
                delay(5000); //  put a bit of a delay so you let go of the 
                digitalWrite(STATUS_LED_PIN, HIGH);
                ESP.reset();
            }

        } else {
            digitalWrite(STATUS_LED_PIN, LOW);
        }

    }

    }

```

probonopd, 2016-12-10:
Thanks for providing this sample codes. I am playing around with 2 devices at the moment and managed to get both into a state at which point I need to connect them to USB to be able to connect to them again... so I'd still appreciate something like this to be the default...

probonopd, 2016-12-10:
Is 

> re-upload the SPIFFS data folder from the example. this will reset it.

also true when uploading the Melvanimate data folder?

sticilface, 2016-12-11:
yes.  any upload of the spiffs folder will wipe the SPIFFS partition on the ESP and reset back to factory settings. 

im a bit curious as to what might be happening if you are able to recreate this lost state again and again.  

Could you maybe enable the debugging by uncommenting this line 
https://github.com/sticilface/ESPmanager/blob/master/src/ESPmanager.h#L58

and show me the debug output when connected by serial... 

Im curious to see where it is getting stuck.

probonopd, 2016-12-11:
Will report back here when I encounter it again.

sticilface, 2017-01-09:
any progress on tracking this down or can i close the issue?

probonopd, 2017-01-11:
Haven't encountered it so far so closing for now. Thanks for checking back.

closed 2017-01-11 by probonopd