sticilface/ESPmanager #16: Documentation for the OTA feature

DanielJean007, 2017-02-12:
First off, great job you've done here. Congrats.
I'm thinking on using this lib with my students in the near future. One thing though is that I wasn't able to use the OTA feature because I couldn't get a hold of the structure your lib expects me to parse. In the /espman/#ota... there's a link to the same page that was supposed to explain such structure. Could you be kind enough to provide a easy-to-follow doc for the OTA feature? I'd really much appreciate it. Thanks and great job again.

sticilface, 2017-02-21:
Hi,

Sorry i didn't see your post for a few days.  Are you referring to the update feature, that you can point the ESP to a URL and it will update the sketch and the SPIFFS files?  and not the OTA that is the stand sketch flashing feature of the ESP8266 Arduino environment.  Just want to be sure I'm answering the right question? 

I've had a look and I agree that the link doesn't work.  I'll make an issue and fix that.  The structure is a JSON object.  `saveto` is the SPIFFS filename, must have the preceding `/`, `location` is the relative path to the manifest file that the file can be found, `md5` is the md5sum of the new file.  This is used to compare against the current file to avoid downloading it if it is unchanged.  The new binary must be called `firmware.bin` and save to = "sketch".  Finally the keys `filecount` tell the updater how many to expect, and `overwrite` means the md5 is ignored and all files are written.  there is an `isurl` option for each file as well, that can be used to specify that the location is a whole url, and not a relative one.  Allows you to download external resources, otherwise the link is treated as relative. 

```json
{  
  "files":[  
    {  
      "saveto":"sketch",
      "location":"/data/firmware.bin",
      "md5":"bbec8986eea6a5836c7446d08c719923"
    },
    {  
      "saveto":"/index.htm.gz",
      "location":"/data/index.htm.gz",
      "md5":"6816935f51673e61f76afd788e457400"
    },
    {  
      "saveto":"/espman/ajax-loader.gif",
      "location":"/data/espman/ajax-loader.gif",
      "md5":"8fd7e719b06cd3f701c791adb62bd7a6"
    },
    {  
      "saveto":"/espman/index.htm.gz",
      "location":"/data/espman/index.htm.gz",
      "md5":"844b0aa50d419bff1fb75b89b12392fe"
    },
    {  
      "saveto":"/espman/settings.json",
      "location":"/data/espman/settings.json",
      "md5":"521186fb21259e7f65f45214916e8a9b"
    },
    {  
      "saveto":"/espman/settingstempl.json",
      "location":"/data/espman/settingstempl.json",
      "md5":"521186fb21259e7f65f45214916e8a9b"
    },
    {  
      "saveto":"/espman/setup.htm.gz",
      "location":"/data/espman/setup.htm.gz",
      "md5":"ca27027b685cab28e5a95bfa30926564"
    }
  ],
  "overwrite":true,
  "filecount":7
}

```

Hope that helps. 
My next version will fix some bugs.  I've written a tool that saves the file directly to flash with zero latency, as I found that as SPIFFS gets full it struggles to find free sectors quick enough and the ESP TCP buffer overflows.  Now it works like a charm with no errors.

DanielJean007, 2017-02-21:
Cool stuff, man. Congrats. 
I was indeed referring to the OTA feature. Correct me if I'm wrong but when you update the sketch doesn't that change the actual firmware on the ESP? I'm interested in a manager for wifi and for flashing OTA just as yours. I'll give it a try shortly.

The explanation and example you gave are helpful.

Just a bit off topic:
Would it be a lot of trouble to make it on demand? I'll explain myself: The whole system is automatic and it can take some action on startup. But what if the user doesn't want to set up an AP right away? Or what if the Internet connection keeps failing and blocks the whole system? Could it be possible to add a routine where your manager checks whether or not the user wants it to work. It'd be something along the lines of what zapu/WiFiManager did, but in that case it's only the Internet manager, yours is much more complete. Here's a link: https://github.com/tzapu/WiFiManager#on-demand-configuration-portal

Thanks again for the time to reply me.

sticilface, 2017-02-21:
I was indeed referring to the OTA feature. Correct me if I'm wrong but when you update the sketch doesn't that change the actual firmware on the ESP?
> Not in this case.  You can push updates via a URL and what ever is in the URL gets updated.  If you do not include a "sketch" + binary then the sketch is not updated.  So you can update only the files.  If you do include a sketch + binary, but it has not changed then it won't get updated.  So only the things that have a different MD5 get downloaded and changed.  so if you have 20 SPIFFS files, and you change one, then only that gets downloaded.  make sense? 

I believe the feature you are asking are already implemented.  on the first page you have a variety of choices for actions to take at boot, and in the case where WiFi fails?

DanielJean007, 2017-02-21:
I see your point.
I saw indeed the actions in the case WiFi fails. I'll upgrade my biometric access system using your work.

Thanks again.

sticilface, 2017-02-21:
cool.  

I use platformio to automate the build process, 
a python script to generate the manifest file
gulp to pipe and compress all the html into one file (much improved performance) 
bash script to upload them all to a local server.

with one command i can update all the ESPs, including SPIFFS files.  You can then trigger an update via MQTT as well / or HTTP i guess, but i then to just poll the update server.

DanielJean007, 2017-02-21:
Hi. It's me again.
Would you have any documentation for errors? I'm getting this one when I try to upload a sketch
[image]
Thanks in advance.

sticilface, 2017-02-21:
I need to make those errors a bit more user friendly. -32 means there is an error with the manifest. -1 is with the http client. Which I think means that it can't connect. If u paste the URL for the file in your browser does it work?

DanielJean007, 2017-02-21:
It does. I used python -m http.server to make a localhost server available. Then, I enter the the address to the json file: http://localhost:8000/ this opens up a directory where there is the json file and the data folder with the firmware.bin inside. Should I do http://localhost:8000/files.json instead?

sticilface, 2017-02-22:
Yes. U should have the exact file name. It can be whatever u want.

sticilface, 2017-02-22:
Can u check with ur mobile as welll. Sometimes these servers only serve local host.

sticilface, 2017-02-22:
Hi. 

I just re read your post and know why it's not working. Localhost on a computer is how the computer identifies itself. So trying to access it from another device will not work. You need to replace it with the computers actual IP address

DanielJean007, 2017-02-22:
Ohh. I see. I'll give it a try. Thanks.

sticilface, 2017-02-24:
Did you get it to work.

DanielJean007, 2017-02-24:
Didn't get it to work just yet. Still working on it, though. I'm gonna use the develop branch for test for sure. Thanks.

thedirtdevil, 2017-07-12:
sorry for kindly asking...can you help me sticilface to show step by step how to solve my problem..my device samsung s3 mini..im stuck with stock 4.1.2 jellybean..help me please how to update/upgrade to kitkat or higher...thank you...

closed 2018-05-01 by sticilface