sticilface/ESPmanager #23: Fatal exception with large list  (20)  of wifi Networks scanned

pit001, 2017-04-26:
During setup i got an: Fatal exception 29(StoreProhibitedCause)
----------------------------------------------------
scandone<\n>
POST[body]: PerformWiFiScan<\n>
JSON length: 2133, heap = 1200<\n>
Begin:<\r><\n>
{<\r><\n>
  "networks": [<\r><\n>
    {<\r><\n>
      "ssid": "xxxxxx",<\r><\n>
      "rssi": -46,<\r><\n>
      "connected": false,<\r><\n>
      "channel": 7,<\r><\n>
      "enc": "WPA2_PSK"
... 19 more
----------------------------------------------------
Exception Decoder:
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 42 results
0x4020decd: AsyncResponseStream::write(unsigned char const*, unsigned int) at C:\...\portable\sketchbook\libraries\ESPAsyncWebServer\src/WebResponses.cpp line 471
0x4021d848: AsyncResponseStream::write(unsigned char) at C:\...\portable\sketchbook\libraries\ESPAsyncWebServer\src/WebResponses.cpp line 471
0x401006f0: malloc at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x4020f067: non-virtual thunk to AsyncResponseStream::write(unsigned char) at ?? line ?
0x4021dad1: ArduinoJson::Internals::JsonWriter::writeRaw(char) at C:\...\portable\sketchbook\libraries\ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp line 153
0x40211fe0: ArduinoJson::Internals::JsonSerializer::serialize(ArduinoJson::JsonObject const&, ArduinoJson::Internals::JsonWriter&) at c:\...\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2058
0x40214999: void ESPmanager::sendJsontoHTTP (ArduinoJson::JsonObject const&, AsyncWebServerRequest*) at c:\...\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2058
0x401008e0: free at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x4020e404: AsyncWebServerResponse::addHeader(String const&, String const&) at C:\...\portable\sketchbook\libraries\ESPAsyncWebServer\src/WebResponses.cpp line 471
0x4021c0a4: operator delete(void*) at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/abi.cpp line 57
0x40207ddc: ESP8266WiFiSTAClass::status() at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 510
0x402172c9: ESPmanager::_HandleDataRequest(AsyncWebServerRequest*) at c:\...\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2058
0x401008b8: realloc at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1714
0x4021b568: String::changeBuffer(unsigned int) at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 720
0x401008e0: free at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x4021dba5: std::_Function_handler    (ESPmanager*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*) at c:\...\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4020f229: std::function ::operator()(AsyncWebServerRequest*) const at C:\...\portable\sketchbook\libraries\ESPAsyncWebServer\src/WebServer.cpp line 79
:  (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at C:\...\portable\sketchbook\libraries\ESPAsyncWebServer\src/WebHandlerImpl.h line 91
0x4020d930: AsyncWebServerRequest::_onData(void*, unsigned int) at C:\...\portable\sketchbook\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 674
0x401008e0: free at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x4010763c: vPortFree at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/heap.c line 18
0x4010020c: _umm_free at C:\...\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4020d9d1: _M_invoke at C:\...\portable\sketchbook\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 674
0x40234665: system_pp_recycle_rx_pkt at ?? line ?
0x4020ab70: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at C:\...\portable\sketchbook\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp line 650
0x4023c549: pbuf_free at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 758
0x4020abc3: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at C:\...\portable\sketchbook\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp line 650
0x40241cbc: tcp_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_in.c line 394 (discriminator 1)
0x4010234b: ppCalFrameTimes at ?? line ?
0x402401bd: ip_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip.c line 559
0x40224b9e: ieee80211_hostap_attach at ?? line ?
0x4021f583: pp_enable_noise_timer at ?? line ?
0x4023f5b9: ethernet_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c line 1379
0x4010542a: lmacTxFrame at ?? line ?
0x40101d96: ppEnqueueRxq at ?? line ?
0x402343e0: ets_timer_handler_isr at ?? line ?
0x40101b63: ppProcessTxQ at ?? line ?
0x40101b9e: ppProcessTxQ at ?? line ?
0x40234063: ets_snprintf at ?? line ?

peter

sticilface, 2017-04-27:
There is a limit of 20 wifi networks, maybe I should reduce it to 10.

sticilface, 2018-05-01:
42bc8d2b555073102fc9f8aa2252cf1001581213

sticilface, 2018-05-01:
https://github.com/sticilface/ESPmanager/commit/42bc8d2b555073102fc9f8aa2252cf1001581213

closed 2018-05-01 by sticilface