sticilface/ESPmanager #28: Setup ESP page is blank

summersab, 2017-10-22:
This project seemed pretty straightforward to flash, is well-documented, and the features sound awesome. However, I can't get the captive portal and initial config to launch. I've tried a variety of browsers, but I just get a blank page. Also, it doesn't seem to redirect well, but browsing to the IP address gets me to what _should_ be the setup page (the page at least has a title element, so that's why I know I got to the right spot). Did I upload something in the wrong order? Am I missing a library?

sticilface, 2017-11-09:
Not entirely sure what the problem is there I'm afraid.  

can you get to the set up page by typing 

http://192.168.4.1/espman/

I know that the captive portal is not super great.  it will only redirect by whole domain, so if you type in an actual page it doesn't work. i remember spending ages on it, but not getting it to work.  

Have you flashed the SPIFFS part to the ESP too?  it needs the page there to work?

sticilface, 2018-05-01:
any news.  feel free to reopen if ongoing problems

closed 2018-05-01 by sticilface